foxglove/studio #4251: Plot with auto x range is very narrow

rgov, 2022-08-24:
- Version: 1.23.0
- OS: macOS
- Data source: ROS1 bag
- ROS distro: Melodic

I am using `rosbag record` with a split option that produces an hourly file. The system runs for a long time, potentially for weeks. When I load one of these slices, a Plot panel that plots a value vs `header.stamp` with auto *x* range looks like this, with all the data squeezed to the right:

[image]

I assume it's setting the minimum *x* limit to the entire lifetime of the ROS master, and not to just what this bag contains. Is there a better solution?

foxymiles, 2022-08-24:
@rgov Can you please check and see if you get the same behavior with this branch? https://github.com/foxglove/studio/pull/4145

rgov, 2022-08-24:
@defunctzombie checked for me and said it looks better :+1:

PR #4145 (2022-08-25): Add manual limits for x axis
**User-Facing Changes**
This adds new plot panel settings to manually specify the min and max X axis values.

**Description**
This adds new settings and new logic to clamp the X axis when both min and max are specified.

It also separates X & Y related settings into their own nodes in the settings sidebar.

[image]

Fixes #3991
Fixes #4251
closed 2022-08-25 by foxymiles