foxglove/studio #4253: [New3D] Cannot load mesh from a URDF using a relative path when fetching over http(s)

jhurliman, 2022-08-24:
**Description**

Relative paths in URDFs to mesh resources fail to load when the URDF is fetched via http(s). Relative paths for other URL schemes such as package:// and file:// have not been tested.

- Version: `main`
- OS: all
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): none (add a custom URDF layer)
- ROS distro (if applicable): N/A

**Steps To Reproduce**

1. Create a URDF file that includes `<mesh filename="./mesh/robot.glb">`
2. Host the URDF at https://myassets.com/robot.urdf and the glTF at https://myassets.com/mesh/robot.glb
3. Create a custom URDF layer in `3D (Beta)` that points at https://myassets.com/robot.urdf
4. Observe the mesh fails to load, but when changing the filename to the absolute URL https://myassets.com/mesh/robot.glb it successfully loads

defunctzombie, 2022-08-25:
Is this something we have ever supported? Or do we only support absolute paths for meshes (i.e. package:// https://) ?

amacneil, 2022-09-01:
Changing to feature because I am not aware of any documentation saying we support this. We should still add it though.

bertaveira, 2022-11-14:
I can confirm that relative paths using the `file://` scheme does not work either