foxglove/studio #4261: [New3D] - Label Scale Size not applied to 3D labels until changed

snosenzo, 2022-08-24:
**Description**
 - if the label scale size is set to a non default and the page is reloaded, the changed number remains in the settings of the 3D panel but it is not applied until the setting is changed.

- Version: 1.23.0-dev
- OS: macos/web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): example dataset

**Steps To Reproduce**

**Expected Behavior**
 - would expect the label in the 3D view to already reflect the label scale in the settings on load

**Actual Behavior**
 - must change label scale to apply it to the 3D view

jtbandes, 2022-08-31:
Can you show a screen recording of the problem? I'm not able to reproduce. I increased the label scale, reloaded, and I see the large labels as soon as data appears.

snosenzo, 2022-09-02:
No longer seeing this, possibly fixed in https://github.com/foxglove/studio/pull/4262

closed 2022-09-02 by snosenzo