foxglove/studio #4263: Support proxy icons on macOS

rgov, 2022-08-24:
Native document-based applications on macOS will use a "proxy icon" that shows where the file is on the filesystem. It would be nice for Studio to support them so it is easier to keep track of where a file is stored, and jump to the containing folder.

[image]

jtbandes, 2022-08-24:
We already do this: https://github.com/foxglove/studio/blob/2fc162bd50c9f301b291178ce4ecbc4cf4aee9eb/desktop/main/index.ts#L173

It looks like a bug that the IterablePlayer does not set its `_filePath`.

defunctzombie, 2022-08-25:
I've long wondered what this file path setting is for or about! I'll take care of this.

jtbandes, 2022-08-25:
This behavior was broken in https://github.com/foxglove/studio/commit/6d0e744df4581b5c97f1497d5c32c762bab34977#diff-8ff00f03f90dedeb147ca074f56cdc45867c552e8383a6c10710c7843ed1f7ceL30

defunctzombie, 2022-08-25:
Upon further investigation this behavior was broken before these changes (likely with the changes to add recents list). The handles we use to work with files in the app don't have a reference to the original file path:

https://github.com/electron/electron/issues/33647

jtbandes, 2022-09-12:
We've explored a direct fix by adding this `path` feature for FileSystemHandle (https://github.com/electron/electron/pull/35604), but the Electron team is not interested in accepting this patch and would like to see a different approach. So this doesn't really fit as a `bug` here — more of a feature request that is blocked on upstream support (https://github.com/electron/electron/issues/33647).