foxglove/studio #4268: Inconsistent `:` menus

amacneil, 2022-08-24:
**Description**
Roman would like the `:` menu to always appear on the right. 

Order should probably be `<error><eyeball><dotdotdot>`

[image]

defunctzombie, 2022-08-25:
We need stories for these different states - also showing more than one item so we can catch issues like this. Elevating to a bug because this is not the intended behavior of the context menu placement.

foxymiles, 2022-08-26:
What should we do if we have two adjacent nodes, both with an error but only one with a context menu? The error indicators will be out of alignment.

defunctzombie, 2022-08-27:
Related question - if a nested node has an error, does the parent also show an error in the collapsed state?

foxymiles, 2022-08-29:
> Related question - if a nested node has an error, does the parent also show an error in the collapsed state?

Currently this is the responsibility of the panel author because I haven't figured out a way to implement this generally that wouldn't compromise performance. Maybe there's a pure CSS solution.

PR #4380 (2022-09-08): Move settings actions after error icon
**User-Facing Changes**
Not worth calling out.

Before

[image]

After

[image]

**Description**
The settings actions are represented as a context menu icon. The typical pattern for context menu icons is to appear at the start or end of the component (or line) as the vertical dot icons are representative of a "more" action.

This change moves the actions icon after the error icon in the NodeEditor.

The error indicator was placed after the actions in https://github.com/foxglove/studio/pull/3384/ which correctly moved the actions after the visibility icon  [... cut]
commit eb7f3200 (2022-09-08): Move settings actions after error icon (#4380)

The settings actions are represented as a context menu icon. The typical
pattern for context menu icons is to appear at the start or end of the
component (or line) as the vertical dot icons are representative of a
"more" action.

This change moves the actions icon after the error icon in the
NodeEditor.

The error indicator was placed after the actions in
https://github.com/foxglove/studio/pull/3384/ which correctly moved the
actions after the visibility icon but also placed them before the error
icon rather than at the end.

Fixes: #4268
closed 2022-09-08 by defunctzombie