foxglove/studio #4279: New 3d Panel: Settings tree jumps on node visiblity toggle

2metres, 2022-08-25:
**Description**
When toggling camera visiblity the Settings Sidebar jumps

- Version: Studio website
- OS: MacOS Chrome
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Example Data
- ROS distro (if applicable):

**Steps To Reproduce**

Expand transforms and topics

Uploading Screen Recording 2022-08-26 at 8.10.40 am.mov…

osaro-imohe, 2022-09-03:
Can I have a video file? I don't have enough data to reproduce this.

[image]

osaro-imohe, 2022-09-04:
I can't reproduce this. It's working fine for me.

snosenzo, 2022-09-06:
I wasn't able to reproduce this on my macbook pro display, only on my I was only able to reproduce on my external display. Maybe it has to do with screen size somehow? @foxymiles any ideas?

foxymiles, 2022-09-06:
I was able to reproduce it but only sporadically after actively clicking a lot of stuff on and off. Maybe something to do with the debounced update to the panel config?

defunctzombie, 2022-09-16:
I've not been able to reproduce this. @2metres are you still able to reproduce this still?

defunctzombie, 2022-09-30:
Haven't heard back. Closing.

closed 2022-09-30 by defunctzombie