foxglove/studio #4284: New3d: Unsetting decay time does not work

defunctzombie, 2022-08-26:
**Description**

* Open a data set with point cloud data and enable the point cloud topic.
* Turn the decay time up to some number of seconds
* Play the data and let points accumulate
* Edit the decay time back to 0

Notice that the decay time does not reset to 0 and we are still seeing accumulated points. I would expect to no longer be accumulating points during playback.

snosenzo, 2022-08-26:
I'm not able to recreate this on studio.foxglove.dev with the example data. is there example data file you saw this with?

defunctzombie, 2022-08-27:
I used the sample data - a local copy of it but it is the same as we currently provide in "explore sample data".

https://user-images.githubusercontent.com/84792/187033970-f6afde78-7e7c-4173-b55a-8ba78d91375f.mov

PR #4345 (2022-09-02): 3D beta: fix some settings bugs
**User-Facing Changes**
Not worth calling out in release notes

**Description**
- Fixes #4284, which occurred because the old settings were being spread in with the new settings, rather than falling back to the defaults.
- Fixes a crash when scrubbing on the decay time field. Adds missing `min` to decay time setting.
- Fixes issues with some settings not updating the UI immediately (because the scene was re-rendering before the update was fully applied).
commit de7f7e90 (2022-09-02): 3D beta: fix some settings bugs (#4345)

**User-Facing Changes**
Not worth calling out in release notes

**Description**
- Fixes #4284, which occurred because the old settings were being spread in with the new settings, rather than falling back to the defaults.
- Fixes a crash when scrubbing on the decay time field. Adds missing `min` to decay time setting.
- Fixes issues with some settings not updating the UI immediately (because the scene was re-rendering before the update was fully applied).
closed 2022-09-02 by jtbandes