foxglove/studio #4298: 3D panel don't support marker with a action DELETEALL and ns nonempty

lizhucheng, 2022-08-29:
**Description**
I got a MarkerArray message which first marker's DELETEALL and id = 0, ns = "LF"(or anything else not ''). Old markers with a lifetime 0 have been deleted. 3D beta panel is ok.

- Version: 1.20.1
- OS:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

**Expected Behavior**
remove markers only if marker's ns value equal the DELETEALL marker.

**Actual Behavior**
removed all old markers in the topic .

jtbandes, 2022-08-29:
Please remember to fill out the requested information in the description – which ROS version are you using? The expected behavior you describe is only documented in ROS 2. https://github.com/ros2/common_interfaces/blob/rolling/visualization_msgs/msg/Marker.msg#L38

Since you said the 3D Beta panel is working, I'm going to close this ticket since we don't plan to make changes to the old 3D panel, and will replace it with the 3D Beta panel very soon.

closed 2022-08-29 by jtbandes

lizhucheng, 2022-08-30:
lt's a msg in ROS 1. I read Marker.msg document. So a DELETEALL action in 3D panel will delete all markers in the topic, right?
What about ImageMarker? ImageMarker.msg has actions include ADD and REMOVE. Can a REMOVE action remove all ImageMarker in the topic? 
Thanks!

jtbandes, 2022-08-30:
Studio does not support `action` for ImageMarker.