foxglove/studio #4306: How do I add pages to this project

Niceyaa, 2022-08-30:
I want to add a custom page in FoxGlove Studio and jump to the main page of the project through the new page, but I add the following code under packages/studio-base/app.tsx:

import { Route, Routes, BrowserRouter as Router, HashRouter } from "react-router-dom";

<Router>
      <Routes>
        <Route path="/data" element={
          <AppConfigurationContext.Provider value={appConfiguration}>
            <ColorSchemeThemeProvider>
              <GlobalCss />
              <CssBaseline>
                <ErrorBoundary>
                  <MaybeLaunchPreference>
                    <MultiProvider providers={providers}>
                      <DocumentTitleAdapter />
                      <SendNotificationToastAdapter />
                      <DndProvider backend={HTML5Backend}>
                        <Suspense fallback={<></>}>
                          <PanelCatalogProvider>
                            <Workspace deepLinks={deepLinks} />
                          </PanelCatalogProvider>
                        </Suspense>
                      </DndProvider>
                    </MultiProvider>
                  </MaybeLaunchPreference>
                </ErrorBoundary>
              </CssBaseline>
            </ColorSchemeThemeProvider>
          </AppConfigurationContext.Provider>
        }></Route>
        <Route path="/" element={
          <div>demo测试</div>
        }>
        </Route>
      </Routes>
    </Router>

Current situation：
Now the http://localhost:8080/ demo测试 can be displayed on the web
At that time, when I type http://localhost:8080/data, prompt always  Cannot GET /data

Now what can I do to make this work, thank you！

defunctzombie, 2022-08-30:
Studio is a single application and does not have multiple pages or a client side router.

closed 2022-08-30 by defunctzombie

Niceyaa, 2022-08-30:
Do you mean that Studio does not support the function of routing new pages @defunctzombie

jtbandes, 2022-08-30:
The issue you are encountering is not a Studio issue; you would need to search the documentation/community for react-router-dom to learn how to fix it.