foxglove/studio #4310: Measurement control overlays on existing widgets

KerstinKeller, 2022-08-30:
**Description**
When you open a measurement, the measurement control opens in the bottom. However, other panels are not resized accordingly, so the control is on top of the other panel.

Resizing the whole Foxglove application fixes the problem.

- Version: 1.24.0
- OS: Windows

**Steps To Reproduce**
- Open Foxglove with a layout
- drop / open a measurement
- See how the measurement control is displayed on top of the widgets

https://user-images.githubusercontent.com/13848742/187442592-43340585-14ba-4eee-8bf5-5bfaaf634e61.mp4

foxymiles, 2022-08-30:
Thanks for the bug report. Have you see this happen with any panels besides the user scripts panel?

KerstinKeller, 2022-08-30:
I just tried the plot panel, which resizes ok. Maybe it's just the user scripts panel....

defunctzombie, 2022-08-30:
Renamed the issue because I am _reasonably_ sure that this is specific to the `user scripts` panel and cutting off the bottom bar in that panel. We'll get this fixed up.

@KerstinKeller as a related note - what you've called the "measurement control" we refer to as the "playback controls".

closed 2022-08-30 by foxymiles