foxglove/studio #4339: Blank 3D image panel when clock times are out of sync

camchambers, 2022-09-01:
**Description**
The 3D image panel in Foxglove Studio blanks when Foxglove Studio is hosted on a robot and a client's clock time is less than the robot's clock time.

Client time and robot time are in sync: 
[image] 

Client time is less than the robot's time: 
[image]

- Version: v1.24.0
- OS: Linux 5.14.0-1050-oem 
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS1

- ROS distro (if applicable): Ubuntu

**Steps To Reproduce**
Run Foxglove Studio on a robot and set your computer's clock back a minute. 

**Expected Behavior**
The 3D image panel renders data appropriately. 

**Actual Behavior**
The 3D image panel renders a blank screen.

defunctzombie, 2022-09-02:
@camchambers Can you please try the same using the `3D (Beta)` panel. We are moving towards deprecating the `3d` panel in favor of the `3D (Beta)` and can look at this issue if it is still impacting the `3D (beta)` panel.

camchambers, 2022-09-06:
Looks like this isn't an issue for the 3D  (beta) panel.

closed 2022-09-06 by camchambers