foxglove/studio #4341: Inconsistent time formatting of start & end times

jtbandes, 2022-09-01:
When viewing this file: [arrows.mcap.zip](https://github.com/foxglove/studio/files/9473795/arrows.mcap.zip)

The sidebar shows the start time in seconds, but the end time as a date:

osaro-imohe, 2022-09-02:
@jtbandes Can I get this?

osaro-imohe, 2022-09-02:
Also, can I get some help setting up Foxglove Studio. I'm running into a generic network error when I try to run Fox Glove Studio web. It seems can't find some UI fonts.  

[image]

foxymiles, 2022-09-06:
Closed by https://github.com/foxglove/studio/pull/4346

closed 2022-09-06 by foxymiles