foxglove/studio #4349: Topic subscription error on Ubuntu but not Windows

adarshrs, 2022-09-05:
- Version: 1.23.0
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1 native
- ROS distro (if applicable): Noetic

**Steps To Reproduce**
1. Launch a launch file on a robot on the same network (with ROS_HOSTNAME properly set).
2. Open Foxglove Studio, select Open Connection > ROS 1. Set ROS_MASTER_URI and ROS_HOSTNAME properly
3. Attempt to plot data from a topic

**Expected Behavior**
Plots data

**Actual Behavior**
Topic subscription error:
```Error: Connection closed unexpectedly
    at TcpConnection._handleClose (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:12:142592)
    at EventEmitter.emit (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1103:12230)
    at file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4745:28763
    at TcpSocketRenderer.messagePort.onmessage (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4745:29246)```

Running the same process from the Windows app works fine. This error only shows up for Live Connections on Ubuntu. Reading rosbags works fine.

defunctzombie, 2022-09-05:
Can you provide more information on how to reproduce this error, the current description is not something we can act on since there could be many unique factors to your setup? In order to effectively debug this issue we need to know more information about your setup, the exact IPs/hostnames, subnets, and whether everything can resolve everything else by hostname on the network. Additional things to check would be whether there are any firewall rules impacting the ubuntu machine but not the windows machine. Are both machines bare-metal or is one of the machines running from a VM? Are any nodes on the robot crashing during the topic subscription?

I also recommend trying the rosbridge connection type. It is more forgiving towards varying networking configurations and firewall rules and requires less setup and tinkering to get connected to a remote robot.

defunctzombie, 2022-09-12:
Closing this out since there's been no further response. This could have been some network configuration or interruption; without more information and debugging there's no further action in Studio.

closed 2022-09-12 by defunctzombie