foxglove/studio #4351: Scaling one synced plot resets manually adjusted Y axis of other synced plots

foxymiles, 2022-09-06:
**Description**
Plot syncing and y axis scale do not behave as expected.

- Version: studio.foxglove.dev
- OS: any
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): nuscenes
- ROS distro (if applicable): -

**Steps To Reproduce**
1. Open nuscenes.
2. Create a new layout with two plots.
3. Manually adjust the vertical scale of one plot by holding down the V key and zooming.
4. Zoom the other plot.

https://user-images.githubusercontent.com/93935560/188710183-782746d3-afee-4cd9-b155-71b92e299f7f.mov

**Expected Behavior**
1. The manually adjusted Y scale should not be overridden.

**Actual Behavior**
Both the X and Y axis of the manually zoomed plot are controlled by zooming in the other plot.

defunctzombie, 2022-10-09:
@foxymiles Were are we with this issue? Are you still looking at it?

foxymiles, 2022-10-10:
Yeah I have a fix that seems to work. I've been trying to find something a bit more elegant but I can put up a PR with what I have if you want to take a look.

PR #4606 (2022-10-21): Improve plot and state transition panel zoom state handling
**User-Facing Changes**
Improves handling of plot and state transition panel zoom state.

**Description**
Change ChartJSManager to respect and not change existing, user manipulated scales except in two cases:
1. When an explicit min/max are specified in settings those take precedence.
2. When we pass an explicit `isBoundsReset` flag in the update message.

The second case is necessary because a min/max of `undefined` is currently interpreted as an intention to let the chart determine its own bounds based on the dataset but this has th [... cut]
commit 9856e7db (2022-10-21): Improve plot and state transition panel zoom state handling (#4606)

Change ChartJSManager to respect and not change existing, user
manipulated scales except in two cases:
1. When an explicit min/max are specified in settings those take
precedence.
2. When we pass an explicit `isBoundsReset` flag in the update message.

The second case is necessary because a min/max of `undefined` is
currently interpreted as an intention to let the chart determine its own
bounds based on the dataset but this has the side effect of clearing any
change in the scale the user has initiated directly, so we want to  [... cut]
closed 2022-10-21 by defunctzombie