foxglove/studio #4368: Plot panel settings series sidebar should save

amacneil, 2022-09-07:
**Description**

- Version:
- OS:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

**Current Experience**

**Desired Experience**

foxymiles, 2022-09-12:
A clarification: the value is being saved on every keystroke. The issue is that it's not leaving the editing state on blur.

defunctzombie, 2022-09-12:
> A clarification: the value is being saved on every keystroke. The issue is that it's not leaving the editing state on blur.

Should we save on every keystroke or go into an edit mode that you can "esc" out of to undo?

foxymiles, 2022-09-12:
> Should we save on every keystroke or go into an edit mode that you can "esc" out of to undo?

Not sure which is better from UX point of view but in general all settings updates are immediate currently. We can certainly handle this differently of course.

amacneil, 2022-09-13:
If we save on every keystroke then why do we need the checkbox thing at all?

I don't particularly care whether we save on every keystroke or not (seems like an implementation detail), but I agree with Roman that it would be nice to hit Esc to cancel/revert the changes.

foxymiles, 2022-09-13:
Currently the purpose of the checkbox is to exit edit mode. I agree it's probably a bit confusing and suggests that your changes aren't saved until you click it.

defunctzombie, 2022-09-13:
A) If we go the immediate save route then lets remove the checkbox since that to me is something I see when I need to confirm my changes not when they've already been saved.

B) If we go the route of matching layout rename workflow then we should avoid saving until the change is accepted so the user can 'esc' to cancel the change.

I am 👍 on us going with A for now since it is visually the least disruptive to the existing workflow and we can explore B as an enhancement pending more feedback.

foxymiles, 2022-09-13:
If we go with A then how should the user exit edit mode? Only on enter?

defunctzombie, 2022-09-13:
Played with this feature some more and I think my only real complaint is that it should exit edit mode of "blur" or loss of focus on the input. If it did that I think it would behave mostly as expected.

foxymiles, 2022-09-13:
@defunctzombie It should be exiting edit mode on blur in that PR. Was that not working for you?

defunctzombie, 2022-09-13:
@foxymiles I had tried on main - didn't notice the PR. I tried the PR 👍 :shipit:

closed 2022-09-15 by foxymiles