foxglove/studio #4373: 3d (beta) panel goes blank when split

defunctzombie, 2022-09-08:
**Description**

Splitting the 3d (beta) panel causes the panel to go blank.

https://user-images.githubusercontent.com/84792/189009801-396a0ea6-3e0a-4b6d-9e77-e64bc61dd247.mov

jtbandes, 2022-09-08:
Guessing this is related to https://github.com/foxglove/studio/pull/3636 – we currently reset panels when the `path.length` changes.

defunctzombie, 2022-09-08:
Agreed. Demoting this to a ux issue so we can remember this and close it out once we have tf preloading.

defunctzombie, 2022-09-09:
Now that we have preloading I would have expected this to work better but when I split the panel I am still not seeing my topics even tho some of them should have loaded as the last latched topic. Do we think this _should_ work now?

defunctzombie, 2022-09-09:
@jtbandes @jhurliman lets look into this now that tf preloading is merged. I suspect there is a problem with how the 3d panel is handling topic data when mounted? Or maybe a problem with how extension panels are provided data when they are remounted.

defunctzombie, 2022-09-10:
I played around with this some more and there are behaviors that I would definitely put more in the bug category - certainly more surprising behavior than I'd think we would have with how we feed messages to panels:

- Working with an mcap file with a single topic `/LIDAR_TOP` still results in the 3d panel going blank when I split it. This behavior surprises me since there is only a single frame to display `LIDAR_TOP` and our player delivers the last message on any subscribed topic. So the 3d panel should have this last message.
- Using the same single topic mcap I can use the "select a point" feature to view the details of a point. There is a "popout" icon in the object selector which opens a raw message panel to view the message details. This results in the 3d panel going blank. Rather unpleasant since I wanted to view the message alongside the 3d view I was just looking at.

Understandably there are some scenes that may not properly reconstruct unless all the state is preserved, but with a single topic single frame data source I would expect the panel to be able to display the scene even through these various splits and changes. Marking this as a bug given this new information about the single frame case.

defunctzombie, 2022-09-12:
John: maybe something in message pipeline?
Jacob: maybe we shouldn't be resetting the panel on splitting?
Roman: notably the raw message panel on split works as expected

PR #4410 (2022-09-13): Fix 3D panel clearing on split
**User-Facing Changes**
Not worth calling out in release notes

**Description**
Fixes #4373
- Reduce unnecessary re-renders when splitting or moving panels (see comment in https://github.com/foxglove/studio/pull/3636#pullrequestreview-1012677374)
- Fix logic in MessagePipeline so latest messages are delivered to new subscribers
commit 96e8e50b (2022-09-13): Fix 3D panel clearing on split (#4410)

**User-Facing Changes**
Not worth calling out in release notes

**Description**
Fixes #4373
- Reduce unnecessary re-renders when splitting or moving panels (see
comment in
https://github.com/foxglove/studio/pull/3636#pullrequestreview-1012677374)
- Fix logic in MessagePipeline so latest messages are delivered to new
subscribers
commit 8f6bcbe8 (2022-09-15): Support more ROS2 message types in live connections and .db3 files (#4408)

This PR updates ROS1/ROS2 dependencies to the latest versions including
the @foxglove/rosmsg-msgs-common change that separately exports ROS1 and
ROS2 types. The Ros2Player has been updated to use ROS2 types, and the
3D panel will toggle between publishing ROS1 or ROS2 messages based on
the data source profile. Fixes #3901

Convert playback controls tooltip to MUI (#4392)

Convert the playback controls tooltip to the MUI tooltip.

Fix 3D panel clearing on split (#4410)

**User-Facing Changes**
Not worth calling out in r [... cut]
closed 2022-09-13 by jtbandes