foxglove/studio #4393: Avoid subscribing to unnecessary topics in 3d panel.

defunctzombie, 2022-09-09:
From a slack converstation:

The 3D panel always subscribes to camera info topics even when no camera image or camera info is turned on. If a user is not visualizing any camera stuff in the 3d panel we don’t want to be streaming the camera information back to the app. Just like we don’t want to be streaming camera images or other random topics that are not being put towards visualizing the scene.

We've agreed that the 3D panel should avoid subscribing to these topics when it does not need to use them.

defunctzombie, 2022-11-17:
@jtbandes I am changing this to a bug. When topics are _off_ in the 3d panel we do not subscribe to them. The camera info topics violate this behavior.

jtbandes, 2022-11-18:
Are we aware of any topics _other_ than cameras which have this problem?

defunctzombie, 2022-11-18:
> Are we aware of any topics _other_ than cameras which have this problem?

I am not aware of any other topics having this problem. I think we've learned in the codebase that the _force_ property for subscriptions is only used for the camera info topics when activating the camera topics.

PR #5044 (2023-01-10): [3D] Replace forced subscription param with shouldSubscribe callback, separate Images and Cameras SceneExtensions
**User-Facing Changes**
- [3D] CameraInfo topics are no longer automatically subscribed
- [3D] Fixes a bug where new CameraInfo messages would not update Image topics in the 3D view

**Description**
The boolean `forced` parameter has been replaced with a `shouldSubscribe()` callback, giving SceneExtensions more flexibility on when subscriptions occur. The Images 
SceneExtension builds its own PinholeCameraModel, decoupling it from the Cameras SceneExtension and fixing a bug with th [... cut]
commit edb7798e (2023-01-10): [3D] Replace forced subscription param with shouldSubscribe callback, separate Images and Cameras SceneExtensions (#5044)

**User-Facing Changes**
- [3D] CameraInfo topics are no longer automatically subscribed
- [3D] Fixes a bug where new CameraInfo messages would not update Image
topics in the 3D view

**Description**
The boolean `forced` parameter has been replaced with a
`shouldSubscribe()` callback, giving SceneExtensions more flexibility on
when subscriptions occur. The Images
SceneExtension builds its own PinholeCameraModel, decoupling it from the
Cameras SceneExtension and fixing a bug [... cut]
closed 2023-01-10 by jhurliman