foxglove/studio #4413: Extend ImageMarker to support text display

windzu, 2022-09-13:
In the image panel, the ImageMarkers supported by foxglove are inherited from ros, but ros did not seem to consider the task of 2d detection at the beginning of its design, so it does not support the display of text.

For 2d detection tasks, category display is also an important requirement. So I hope you can add an extension to the text display on the basis of ImageMarkers

defunctzombie, 2022-09-13:
@jtbandes Do you think this is a good candidate for the BYOD epic via a foxglove.TextAnnotation message next to our existing ImageAnnotation messages?

jtbandes, 2022-09-13:
Could be a good idea, time permitting.

jtbandes, 2022-09-13:
Duplicate of https://github.com/foxglove/schemas/issues/24

closed 2022-09-13 by jtbandes