foxglove/studio #4431: Developer console logs are very noisy by default

amacneil, 2022-09-15:
**Description**
Developer console should not show info-level logs by default, only errors + warnings.

[image]

User impact is that this makes extension development difficult due to all the noise.

jtbandes, 2022-09-15:
> should not show info-level logs by default, only errors + warnings

Seems like this isn't something we have control over? Did you mean that we should downgrade some of our 'info' logs to 'debug' logs?

defunctzombie, 2022-09-15:
My plan is to default disable all logs except _errors_ and _warnings_ in production. The new `Studio Logs Settings` panel will have a toggle to turn on the other log levels in production for inspection and debugging.

> Seems like this isn't something we have control over? Did you mean that we should downgrade some of our 'info' logs to 'debug' logs?

We do have control over this. All our logging happens through @foxglove/log and I will add a way for that to filter out specific levels. It already has a way to disable specific log categories.

PR #4439 (2022-09-16): Set minimum log level to _warn_ for production builds

**User-Facing Changes**
None.

**Description**
Add a new dropdown to the Studio Logs Settings sidebar (sidebar is under a feature flag), which allows setting the global log level.

The level sets the minimum severity for writing logs to console.

[image]

This change updates the default minimum severity to "warn" for production builds.

Fixes: #4431
commit db102fef (2022-09-16): Set minimum log level to _warn_ for production builds (#4439)

Add a new dropdown to the Studio Logs Settings sidebar which allows setting the global log level. The level setting sets the minimum severity for writing logs to console.

This change updates the default minimum severity to "warn" for production builds.

Fixes: #4431
closed 2022-09-16 by defunctzombie