foxglove/studio #4446: User Scripts do not run if foxglove opened by shareable link

RileyEv, 2022-09-20:
**Description**

When attempting to open remote rosbags with a shareable link and a saved layout with user scripts, the user scripts do not run. The visualisations still work, it just affects anything that depends on user scripts. You can get the scripts to run by deleting and re-importing them. If opening foxglove without a direct link and then opening a remote rosbag from a url, there are no issues. 

- Version: 1.25.0
- OS: I've been able to reproduce this on the web version and on local versions on Linux. 
- Data source: Remote rosbag opened with shareable link.

**Steps To Reproduce**

1. Add the example layout from below and ensure that it is the layout that opens by default.
2. Open this link: `https://studio.foxglove.dev/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fassets.foxglove.dev%2FnuScenes-v1.0-mini-scene-0061.bag`. This will open the sample foxglove data in the browser, the layout should then open.
3. User scripts do not run. I get the error message in the user scripts window `Input "/gps" is not yet available InputTopicsChecker`

**Expected Behavior**
In the ros message inspector panel on the right, I'd expect to see `"Hello World"` produced at the same Hz as the `/gps` message.

**Actual Behavior**
The panel shows nothing as the message is not being produced by the user scripts.

**Example Layout**
(sorry it wouldn't let me upload json! :cry:)
```json
{
  "configById": {
    "NodePlayground!1nieorr": {
      "selectedNodeId": "394d2b33-eba0-4d22-baac-3615ee5f0d29",
      "autoFormatOnSave": true
    },
    "RawMessages!25axhjx": {
      "diffEnabled": false,
      "diffMethod": "custom",
      "diffTopicPath": "",
      "showFullMessageForDiff": false,
      "topicPath": "/studio_script/output_topic"
    },
    "3D!1z1k64b": {
      "cameraState": {
        "distance": 19.999999999999982,
        "perspective": true,
        "phi": 59.99999999999997,
        "target": [
          0,
          0,
          0
        ],
        "targetOffset": [
          0.027861423906154804,
          0.027861423906154807,
          0
        ],
        "targetOrientation": [
          0,
          0,
          0,
          1
        ],
        "thetaOffset": 44.99999999999999,
        "fovy": 45,
        "near": 0.01,
        "far": 5000
      },
      "followMode": "follow-pose",
      "scene": {},
      "transforms": {},
      "topics": {
        "/LIDAR_TOP": {
          "visible": true,
          "colorField": "intensity",
          "colorMode": "colormap",
          "colorMap": "turbo"
        }
      },
      "layers": {},
      "publish": {
        "type": "point",
        "poseTopic": "/move_base_simple/goal",
        "pointTopic": "/clicked_point",
        "poseEstimateTopic": "/initialpose",
        "poseEstimateXDeviation": 0.5,
        "poseEstimateYDeviation": 0.5,
        "poseEstimateThetaDeviation": 0.26179939
      }
    }
  },
  "globalVariables": {},
  "userNodes": {
    "394d2b33-eba0-4d22-baac-3615ee5f0d29": {
      "sourceCode": "// The ./types module provides helper types for your Input events and messages.\nimport { Input, Message } from \"./types\";\n\n// Your script can output well-known message types, any of your custom message types, or\n// complete custom message types.\n//\n// Use `Message` to access your data source types or well-known types:\n// type Twist = Message<\"geometry_msgs/Twist\">;\n//\n// Conventionally, it's common to make a _type alias_ for your script's output type\n// and use that type name as the return type for your script function.\n// Here we've called the type `Output` but you can pick any type name.\ntype Output = {\n  hello: string;\n};\n\n// These are the topics your script \"subscribes\" to. Studio will invoke your script function\n// when any message is received on one of these topics.\nexport const inputs = [\"/gps\"];\n\n// Any output your script produces is \"published\" to this topic. Published messages are only visible within Studio, not to your original data source.\nexport const output = \"/studio_script/output_topic\";\n\n// This function is called with messages from your input topics.\n// The first argument is an event with the topic, receive time, and message.\n// Use the `Input<...>` helper to get the correct event type for your input topic messages.\nexport default function script(event: Input<\"/gps\">): Output {\n  return {\n    hello: \"world!\",\n  };\n};",
      "name": "394d2b33"
    }
  },
  "linkedGlobalVariables": [],
  "playbackConfig": {
    "speed": 1
  },
  "layout": {
    "direction": "row",
    "first": "3D!1z1k64b",
    "second": {
      "first": "NodePlayground!1nieorr",
      "second": "RawMessages!25axhjx",
      "direction": "row"
    }
  }
}
```

jtbandes, 2022-09-20:
Could you give it a try with this preview build of the app? https://vercel.com/foxglove/studio/CFzNMTKVXC9bjkH44RE9j8iQjqEz

RileyEv, 2022-09-21:
I've tested that deployment and deployed your branch locally, everything looks to be working now. Thank you for the quick fix!

closed 2022-09-21 by jtbandes