foxglove/studio #4447: "Add event" icon is misplaced on narrow screen

defunctzombie, 2022-09-20:
Open a data platform data source, open the sidebar, and make the app narrow.

[image]

I don't know how best to hide the icons when the playback controls are narrow but I think the current display looks "broken" rather than "adapting" to the size.

foxymiles, 2022-09-21:
This is going to be hard to fix unless we're willing to use a container query. It's not the same problem as a normal responsive query since the @media queries are going to return a large screen but the actual window is narrow. That said, using a narrow window on a large screen seems like a pretty niche use case.

foxymiles, 2022-09-21:
For example, on a device with a narrow screen the layout is reasonable IMO:

[image]