foxglove/studio #4453: Currently visible layout is not distinguished from other layouts when multi-selecting

jtbandes, 2022-09-20:
**Description**
- Version: latest web
- OS: macos

**Steps To Reproduce**
Select multiple layouts in the sidebar by cmd-clicking.

**Current Experience**
Since all selected layouts appear identical, it is unclear which layout is **currently** being viewed on the right-hand side of the screen.
[image]

**Desired Experience**
Some indication of which layout is actually being viewed

osaro-imohe, 2022-10-06:
Quick question:
1) What are those purple dots opposite the layout name for?
2) Can Studio use that to identify the selected layout?

I'm thinking something along the lines of showing the purple dot only for the selected layout

jtbandes, 2022-10-06:
The purple dot means the layout has been modified since it was last saved.

osaro-imohe, 2022-10-06:
Ah, I see. What if Studio uses a darker shade of purple for the selected layout? Thoughts @amacneil @2metres

amacneil, 2022-10-07:
I'm not convinced we need to solve this problem. The same thing happens in vscode for example. You are only multi-selecting to take an action on those layouts so it doesn't seem critical to know which layout is active?

jtbandes, 2022-10-07:
VS Code does something a bit different. It distinguishes the most recently clicked file when multi-selecting:

[image]

For VS Code highlighting the currently _open_ file in the editor when multi-selecting isn't a big deal, because the open file is already displayed somewhere, namely in the actual tab:
[image]

Plus they have a whole pane dedicated to showing the "open editors".
[image]

IMO a darker purple as background would be nice, but will let @2metres weigh in