foxglove/studio #4470: Use `queue_length` for rosbridge subscriptions

achim-k, 2022-09-22:
When subscribing to a rosbridge topic, studio currently does not set the [queue_length](https://github.com/RobotWebTools/rosbridge_suite/blob/a4987221bfaa4d6fe74f1ab0464832b71393935a/ROSBRIDGE_PROTOCOL.md#334-subscribe) parameter.

https://github.com/foxglove/studio/blob/22f6bf52cb1f8e093d1d8f6eb8c588209ff6033c/packages/studio-base/src/players/RosbridgePlayer.ts#L460-L464

When the `queue_length` is 0 (default), the `rosbridge_server`  uses the default [MessageHandler](https://github.com/RobotWebTools/rosbridge_suite/blob/d6c3f22ee414e259e11d4f6df49c9c5a81129e38/rosbridge_library/src/rosbridge_library/internal/subscription_modifiers.py#L47) which processes incoming messages in the `rospy.Subscriber` thread. If incoming messages can't be processed and written on the websocket fast enough (e.g. because many high bandwidth topics are subscribed), the subscriber thread is not able to keep up with incoming messages and the publisher will eventually queue them up. This will cause the subscriber to receive messages with a delay. From [ROS wiki]( http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers#Choosing_a_good_queue_size): 
> But if your queue is much larger than it needs to be that will queue up a lot of messages if a subscriber is lagging behind. This might lead to messages arriving with large latency since all messages will be delivered in FIFO order to the subscriber once it catches up.

If a `queue_length > 0` is set, the `rosbridge_server` uses the [QueueMessageHandler](https://github.com/RobotWebTools/rosbridge_suite/blob/d6c3f22ee414e259e11d4f6df49c9c5a81129e38/rosbridge_library/src/rosbridge_library/internal/subscription_modifiers.py#L105) which handles message processing in its own thread. The `rospy.Subscriber` thread can keep up with incoming messages as it is only responsible for receiving incoming messages and pushing them on the message handler's queue.

---

**TL;DR**: For studio I would propose to use a `queue_length` of 1 by default, as this would avoid receiving messages with a large delay. Even better would be to allow the user to choose the `queue_length` by themselves (globally or on a topic level)

defunctzombie, 2022-10-21:
@achim-k From our recent conversations it sounds like you no longer think this would be a good change and that we should keep the `queue_length` value `0` to avoid spawning more threads on the rosbridge side? If we want to keep this at 0 lets close out this feature ticket by adding an explicit `queue_length: 0` argument with a comment explaining why we've selected 0 so our future selves have this context.

achim-k, 2022-10-24:
Correct, in my tests where `rosbridge_server` was bridging many topics (including heavy point cloud topics), setting a `queue_size` > 0 resulted in a slightly degraded performance (in terms of message throughput). 

However, a `queue_size` of > 0 might help to reduce message delay in some cases, e.g. when the `rosbridge_server` is not under high load and only subscribing to few topics. In #2873 there was a request to allow setting the `queue_length` on a topic level. I think a good solution for now would be to allow the user to set the `queue_size` (default is 0) when configuring the rosbridge datasource. What do you think?

defunctzombie, 2022-10-27:
> I think a good solution for now would be to allow the user to set the `queue_size` (default is 0) when configuring the rosbridge datasource. What do you think?

Technically I think this makes sense. For most users I don't imagine they'll have enough context to decide on what the correct tradeoff is and whether they should use 0, 1, 2+?. The name is also a bit ambiguous since a queue size of 0 might make you think thats actually what you want to reduce delay.

I think given the confusion with this field name and impact I would favor not changing anything here for now and keeping this in mind as we work on our higher performing bridge.