foxglove/studio #4473: Encoding Error ,cannot support Chinese character in RawMessage panel

hql420621, 2022-09-23:
**Description**

- Version: 1/24
- OS: ubuntu20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**
+ step1 add two panel ,Raw message  and Publish panel
+ step 2  choose a topic in Publish panel and change the value to Chinese Character
+ step3 subscribe the  same topic in Raw message,

**Expected Behavior**
the Chinese Character encoding show correct

**Actual Behavior**
error encoding
[image]

jtbandes, 2022-09-23:
Please fill in information about your data source. This is likely a problem with rosbridge.

hql420621, 2022-09-23:
there is output of "rostopic echo ", it seems the rosbridge translate "中国" to "\u4E2D\u56FD" encode
[image]

closed 2022-10-05 by defunctzombie

jtbandes, 2022-10-05:
Thank you for the report! The issue should be fixed now on studio.foxglove.dev and will be published in the next release.

PR #4567 (2022-10-05): update @foxglove/rosmsg-serialization
**User-Facing Changes**
Fixed an issue where ROS messages containing non-ASCII strings would sometimes be incorrectly decoded.

**Description**
Fixes #4473
[image]
commit 6e23fbf6 (2022-10-05): update @foxglove/rosmsg-serialization (#4567)

Fixed an issue where ROS messages containing non-ASCII strings would
sometimes be incorrectly decoded.

Fixes #4473