foxglove/studio #4478: Unintended placement of UI elements in panel full-screen mode

esthersweon, 2022-09-25:
**Description**
- Strange and likely unintentional placement of UI components in panel full-screen mode (see playback bar and panel settings below):

[image]

- Version: 1.25
- OS: MacOS Monterey 12.0
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Sample data
- ROS distro (if applicable): N/A

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2022-09-26:
Related to this - I'm tempted to say that maybe "full screen" should mean "panel area" rather than sidebar? I think a useful workflow is to "full screen" a panel to configure it. For this we need sidebar visibility.

foxymiles, 2022-09-26:
We should add a story for this too as part of the fix.

jhurliman, 2022-09-27:
> Related to this - I'm tempted to say that maybe "full screen" should mean "panel area" rather than sidebar? I think a useful workflow is to "full screen" a panel to configure it. For this we need sidebar visibility.

As long as there is a "full full screen" option available.

closed 2022-09-28 by jtbandes