foxglove/studio #4484: Cannot toggle visibility for `/robot_description` topic

jhurliman, 2022-09-26:
**Description**

From a user report, playing `/robot_description` over a ROS2 native connection shows the topic in the 3D panel but the visibility icon does nothing when clicked (it does not toggle from hidden to visible).

https://user-images.githubusercontent.com/195374/192342352-ddab1832-36bd-46c8-942e-9c030bead5cd.mp4

defunctzombie, 2022-09-28:
@jhurliman Do we have an mcap file or bag file with a `/robot_description` topic we can use to reproduce this? Or is this specific to live connections?

jhurliman, 2022-09-28:
@defunctzombie I'm not aware of any recordings with that topic. The user report was about a live connection.

defunctzombie, 2022-10-24:
John: We support receiving urdf over a parameter and a topic. If you receive it as a parameter we show it in the UI as a "topic" but that topic can't be toggled because there is no way to unsubscribe from a parameter. This results in "broken" visibility toggle because there is nothing to toggle.

For the parameter we should show the URDF in some other location and not in the topic list.

amacneil, 2022-11-02:
Product review feedback:

- Specifically look for `/robot_description` parameter, and if it exists just show it as a Topic named `/robot_description (parameter)`.
- Fix the bug where visibility toggle is not working.

closed 2022-11-09 by jtbandes