foxglove/studio #4497: App crashes when pasting color value into Gradient picker

defunctzombie, 2022-09-28:
The app crashes when pasting a color value into the Gradient picker (settings editor)

1. Open the `Colors` storybook: https://603ec8bf7908b500231841e2-ftdjtiqpey.chromatic.com/?path=/story/components-settingstreeeditor--colors
2. Click on the first gradient color box
3. Copy the hex color value
4. Click on the second gradient color box
5. Paste the color value

https://user-images.githubusercontent.com/84792/192801718-1cdb6294-c1ee-4788-81b6-e917a8c56dde.mov

```
Minified MUI error #9; visit https://mui.com/production-error/?code=9&args[]=000000ff for the full message.
```

jtbandes, 2022-09-28:
You can also reproduce this by simply typing 000000ff into the box.

2metres, 2022-09-28:
I think the error is coming from `getContrastText()` we need to sanitize the color before we pass it to the swatch

PR #4510 (2022-10-03): Safely parse color values in the ColorSwatch component
**User-Facing Changes**
None

**Description**
Use tinycolor to parse the incoming color value before passing it on to MUI's color utility to avoid throwing an exception on an invalid color.

Fixes #4497
closed 2022-10-03 by foxymiles