foxglove/studio #4509: Extension in organisation disappears when foxglove opens without internet connection

bertaveira, 2022-09-29:
**Description**
I have noticed that sometimes the extension that we have installed in our organisation does not show up on the desktop version of foxglove. This seems to happen sometimes when we open foxglove without internet on a computer that previously had used the extension and is still logged in.

The quick fix is always to connect to a hotspot, open foxglove and then disconnect the hotspot.

- Version: 1.26
- OS: MacOS

foxymiles, 2022-09-29:
When you say it doesn't show up do you mean you get this error message or that you're unable to use panels from that extension in layouts?

[image]

bertaveira, 2022-09-29:
No I get an error like this:
[image]

And on the panel sidebar the extension panel is no longer there. Local installed extensions are there and work but organisations extensions not

foxymiles, 2022-09-29:
Ok thanks for the clarification. When you don't have a connection what do you see in the extension sidebar?

bertaveira, 2022-09-29:
Oh sorry. Yes I see the error you shared in the screenshot

foxymiles, 2022-09-29:
I'm trying this to reproduce:

1. Start a new session using a layout with a panel from an org extension.
2. Turn off the network on my machine.
3. Reload.

In this case the layout and panel still load ok.

When you say you open it without internet, does that mean that you have a live network connection but it's not able to talk to the internet or is the network completely disabled?

bertaveira, 2022-09-29:
If I open it with internet it load the extension properly. Then if I disable wifi and open again it still works. Then a day later I open again still without internet and the extension is gone. It does not happen every time but it seems that after a while the extension gets removed and unless it has internet it is no longer able to load.

foxymiles, 2022-09-29:
Ok thanks for the extra details. I'm looking into it now.

bertaveira, 2022-09-30:
Ok today something even weirder happened. I opened foxglove with internet connection and it loaded the extension fine for 5 seconds and then it refreshed and the extension and the panel was gone. Restarting foxglove fixed it.

bertaveira, 2022-09-30:
Ok it got even weirder. Now it recognizes the extension properly but it can't find the panel. When I remove the panel or replace it the entire tab layout breaks:

https://user-images.githubusercontent.com/9644729/193250425-b3ca2c04-b8f6-40de-a39b-837ba69f7364.mov

bertaveira, 2022-09-30:
Restarting again the extension is gone. Again and it is back. Very weird behaviour

foxymiles, 2022-09-30:
Sorry about the strange behavior. I'm still investigating and trying to reproduce it here.

closed 2022-09-30 by foxymiles