foxglove/studio #4512: Apt updates not working for 1.26.0

RileyEv, 2022-09-29:
I feel like something may be up with apt updates for `1.26.0`. Multiple people who installed `1.25.0` or lower are not seeing `1.26.0` even after running apt update.  They can only see these versions:
```
foxglove-studio/stable,now 1.25.0 amd64 [installed]
foxglove-studio/stable 1.24.0 amd64
foxglove-studio/stable 1.23.0 amd64
foxglove-studio/stable 1.22.0 amd64
foxglove-studio/stable 1.21.0 amd64
foxglove-studio/stable 1.25.0 arm64
foxglove-studio/stable 1.24.0 arm64
foxglove-studio/stable 1.23.0 arm64
foxglove-studio/stable 1.22.0 arm64
foxglove-studio/stable 1.21.0 arm64
```

After installing the `1.26.0` deb I get
```
apt list -a foxglove-studio 
Listing... Done
foxglove-studio/now 1.26.0 amd64 [installed,local]
foxglove-studio/stable 1.25.0 amd64
foxglove-studio/stable 1.24.0 amd64
foxglove-studio/stable 1.23.0 amd64
foxglove-studio/stable 1.22.0 amd64
foxglove-studio/stable 1.21.0 amd64

foxglove-studio/stable 1.25.0 arm64
foxglove-studio/stable 1.24.0 arm64
foxglove-studio/stable 1.23.0 arm64
foxglove-studio/stable 1.22.0 arm64
foxglove-studio/stable 1.21.0 arm64
```
But `1.26.0` does not seem match the pattern.

defunctzombie, 2022-09-29:
The apt release failed for v1.26.0. I am re-running the release now.

defunctzombie, 2022-09-29:
@RileyEv The apt release is complete. Please try updating via apt. If that works please close out this issue.

RileyEv, 2022-09-29:
Yes I can confirm its working now thanks

closed 2022-09-29 by RileyEv