foxglove/studio #4518: Cannot run yarn, repeated corepack not enabled error

jhurliman, 2022-09-30:
**Description**

```
➜  studio git:(main) yarn                   
This repository uses corepack. Enable corepack by running `corepack enable`
Learn more at: https://nodejs.org/api/corepack.html
➜  studio git:(main) corepack enable
➜  studio git:(main) yarn
This repository uses corepack. Enable corepack by running `corepack enable`
Learn more at: https://nodejs.org/api/corepack.html
➜  studio git:(main) corepack enable
➜  studio git:(main) yarn           
This repository uses corepack. Enable corepack by running `corepack enable`
Learn more at: https://nodejs.org/api/corepack.html
```

- Version: main
- OS: MacOS

jtbandes, 2022-09-30:
I can't reproduce this. I tried:

```
corepack disable
npm i -g yarn
yarn  # corepack error
corepack enable
yarn  # works
```

defunctzombie, 2022-09-30:
Works for me as intended

```
(base) ➜  studio git:(main) node --version
v16.17.0
(base) ➜  studio git:(main) yarn --version
3.1.0
(base) ➜  studio git:(main) which yarn
/Users/roman/n/bin/yarn
(base) ➜  studio git:(main) which node
/Users/roman/n/bin/node
(base) ➜  studio git:(main) corepack disable
(base) ➜  studio git:(main) yarn
zsh: command not found: yarn
(base) ➜  studio git:(main) corepack enable
(base) ➜  studio git:(main) yarn
➤ YN0000: ┌ Resolution step
➤ YN0002: │ @fluentui/react-icons-mdl2@npm:1.2.14 [6eb1c] doesn't provide @types/react-dom (p41951), requested by @fluentui/react-icon-provider
```

I would check on how you installed yarn or what yarn is actually running.

defunctzombie, 2022-10-02:
Issue was with John's computer.

closed 2022-10-02 by defunctzombie

jaguardo, 2023-04-10:
have the same issue... assume yarn has to be removed globally to solve @jhurliman ?