foxglove/studio #4526: Data looks wrong when too many TFs are preloaded

jtbandes, 2022-09-30:
**Description**
3D user data appears in the wrong position when too many TFs are preloaded.

**Steps To Reproduce**
- Load dataset with large number of TF messages (significantly over the current limit of 10,000)

**Expected Behavior**
- Data appears in the correct position regardless of current time

**Actual Behavior**
- Preloaded TFs from the future cause old TFs from near the playhead to be deleted, so the wrong TFs are used to display current data.

jtbandes, 2022-09-30:
A few ideas for what we can do:

- #4525 
- Display a warning/error to the user when we reach the limit on preloaded TFs
- Add a setting to configure the preloading capacity
- Rework TF caching to allow preloading _some_ TFs up to a limit, but still display correctly at the current timestamp

jtbandes, 2022-10-07:
@defunctzombie Any update on whether you feel comfortable closing this now?

defunctzombie, 2022-10-07:
> @defunctzombie Any update on whether you feel comfortable closing this now?

I've not come back to checking it yet. Got sidetracked with the worker stuff. Its on my radar but is likely a weekend early next week thing for me.

defunctzombie, 2022-10-09:
I've written up a new issue describing my visualization problem here: https://github.com/foxglove/studio/issues/4605 This new issue does not relate to too many TFs preloaded.

jhurliman, 2022-10-10:
The bug aspect of "Data looks wrong when too many TFs are preloaded" has been fixed in https://github.com/foxglove/studio/pull/4534. A few different ideas are floating around for improving TF handling for large amounts of transforms, and for improving TF preloading. We could either open individual tickets for specific approaches, or feature request meta-tickets for large TF volumes and TF preloading (related but separate issues). I will close this bug report ticket now that the user is informed when data loss occurs and a partial mitigation is in place by disabling TF preloading.

closed 2022-10-10 by jhurliman