foxglove/studio #4549: Upload rosbag to website

hazcoper, 2022-10-04:
**Description**

We are struggling to get some rosbags into the data plataform. We are getting errors in transcoding: "Processing failure: failed to transcode import: transcoding error: failed to parse opcode: key op not found." We have reindexed the bag.

the following google drive link will take you to the rosbag that we are struggling to upload: 
https://drive.google.com/file/d/1p7J5L403AJFhhjrpDceK0qrE7SBDSwRb/view?usp=sharing 

- Version:
- OS:  Ubuntu 18.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbag

**Error**
Processing failure: failed to transcode import: transcoding error: failed to parse opcode: key op not found

wkalt, 2022-10-04:
@hazcoper thanks. This is caused by a bug in the rosbag reindex command, which is ticketed and described here: https://github.com/ros/ros_comm/issues/2282

@emersonknapp has a fix for that here: https://github.com/ros/ros_comm/pull/2286

As a temporary workaround, you can run your bags through this script, after running rosbag reindex on them: https://gist.github.com/janpaul123/deaa92338d5e8309ef7aa7a55d625152

defunctzombie, 2022-10-04:
Thanks for reporting this. I'm going to close this out since the underlying issue is that the bag file is still malformed (even after re-index). The fixes are in progress for ros_comm and a workaround script is available to fix your bag file.

If the above script or ros_comm fixes do not produce working files for you please let us know!

closed 2022-10-04 by defunctzombie