foxglove/studio #4552: Mesh Import Discrepancy Between 3D and 3D (Legacy)

cody-sheff, 2022-10-04:
Currently there is a mismatch between the orientation that meshes are imported in the 3D tab, and the 3D (Legacy) tab. Attached is a .dae file that demonstrates the issue; when imported in the 3D and 3D (Legacy) tab, there seems to be a 90-degree difference about the z-axis between them.

Sample mesh to demonstrate the issue: [ybase.zip](https://github.com/foxglove/mcap/files/9708097/ybase.zip)

jtbandes, 2022-10-04:
After creating a sample mcap file to show this mesh in the 3d panel, it looks like this:

[mesh.mcap.zip](https://github.com/foxglove/studio/files/9710472/mesh.mcap.zip)

[image]

Which is the same as Blender: 
[image]

So this is behaving as intended.

closed 2022-10-04 by jtbandes

horverno, 2022-10-21:
I can confirm the 90 deg roll error in Foxglove Studio 1.29.0 compared to rviz.

[image]

[image]

jtbandes, 2022-10-21:
Please try this new setting: https://github.com/foxglove/studio/pull/4577

horverno, 2022-10-24:
> Please try this new setting: #4577

Thank you @jtbandes New3D panel>>Settings>>Scene>>Mesh up axis set to `Z-up` solves the problem :+1: 

[image]