foxglove/studio #4592: JointTrajectory time_of_start is not copied from Publish panel

jos-technion, 2022-10-07:
**Description**
I'm using the Publish panel to publish a jointTrajectory msg.
After filling the json, the actual msg sent to ROS doesn't include the header or the time_of_start, now the header i understand as it is filled automatically (though If i filled it, i expect it to be sent as is), but the time_from_start is sent as 0 seconds, 0 nsec, and it is causing a problem

- Version: 1.27
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): live ROS 1
- ROS distro (if applicable): noetic

**Steps To Reproduce**

Publish to joint trajectory, and look at the published msg
UI Publish JSON:
{
  "header": {
    "seq": 0,
    "stamp": {
      "sec": 0,
      "nsec": 0
    },
    "frame_id": ""
  },
  "joint_names": [
    "camera_x_joint"
  ],
  "points": [
    {
      "positions": [
        0.02
      ],
      "velocities": [
        0
      ],
      "accelerations": [
        0
      ],
      "effort": [
        0
      ],
      "time_from_start": {
        "sec": 0,
        "nsec": 5
      }
    }
  ]
}

**Expected Behavior**
header: 
  seq: 5
  stamp: 
    secs: 1665174559
    nsecs: 449193477
  frame_id: ''
joint_names: 
  - camera_x_joint
points: 
  - 
    positions: [0.02]
    velocities: [0.0]
    accelerations: [0.0]
    effort: [0.0]
    time_from_start: 
      secs: 0
      nsecs:         5

**Actual Behavior**
header: 
  seq: 5
  stamp: 
    secs: 1665174559
    nsecs: 449193477
  frame_id: ''
joint_names: 
  - camera_x_joint
points: 
  - 
    positions: [0.02]
    velocities: [0.0]
    accelerations: [0.0]
    effort: [0.0]
    time_from_start: 
      secs: 0
      nsecs:         0

defunctzombie, 2022-11-01:
@jos-technion I opened Studio (v1.29.0), added a Publish panel for `trajectory_msgs/JointTrajectory` and changed the `time_from_start` to some non-zero value. When I do `rostopic echo` on my published topic I am able to receive the message as expected with the non-zero values I entered.

I am not able to reproduce the issue you describe. If you could provide a more specific set of steps or a video showing the problem maybe there is a different sequence of steps I need to take to reproduce the issue.

defunctzombie, 2022-11-08:
Since I have been unable to reproduce this with the information provided and there's been no follow-up I am closing this as not-planned. If you can provide more information how to reproduce I am happy to revisit.

closed 2022-11-08 by defunctzombie