foxglove/studio #4598: publish panel can not publish the topic in version 1.27

xiaolaji0426, 2022-10-08:
**Description**
I can not publish the topic through the publish-panel, but it works in version 1.19

- Version:1.27
- OS:Ubuntu 18
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

**Expected Behavior**
When I edit the topic to disarm the drone ,(and i swear I didn't get the topic wrong as well its datatype!) the console of drone should show this change.

**Actual Behavior**
Nothing happened/(ㄒoㄒ)/~~
I think it could be some bug in different version, please help me

foxymiles, 2022-10-10:
What connection type are you using here? ROS or Rosbrige or something else? Please provide as many details about your environment as you can.

xiaolaji0426, 2022-10-12:
I did use ROS. Actually, studio can subscribe my custom-topic, and my subject is based on Gazebo and PX4.

Here are my topic and datatype:
topic:/xtdrone/iris_0/cmd
datatype:std_msgs/String

By the way, I'm sure that this topic works because I succeeded in publishing by pyros.
Help me, pls, this problem has been bothering me for weeks.🙁

foxymiles, 2022-10-12:
What are you specifying for the ROS_HOSTNAME in this connection dialog?

[image]

foxymiles, 2022-10-12:
Part of the problem is here:

https://github.com/foxglove/studio/blob/c533084b234fd5f438d19f1abd6f0ca81acf858b/packages/studio-base/src/players/Ros1Player.ts#L121-L125

The hostname we pass to `listen` here is the IP address we later respond with when subscribers try to subscribe to our topic. So if this is undefined or `localhost` or `::` or anything like that subscriptions will fail.

foxymiles, 2022-10-12:
Also unfortunately it looks like when we respond to the ROS `requestTopic` call we need to return an IP address and not a hostname:

https://github.com/robotika/husky/blob/bbe57fc80a9d2e99c027a57390af07df064e826e/ros/node.py#L119-L123

PR #4631 (2022-10-13): Revert ROS1 player to bind on ROS_HOSTNAME
**User-Facing Changes**
Fixes an issue in publishing to a ROS1 system from studio.

**Description**
Fixes regression from https://github.com/foxglove/studio/pull/4400.

The issue is that the `hostname` we pass to the TCP server `listen` call is returned in the `requestTopic` call:

https://github.com/foxglove/ros1/blob/73839e31c678749042651ff852fa39a1e4634f29/src/RosFollower.ts#L175-L198

This needs to be an actual IP address that other subscribers can reach so returning something like `localhost` or `0.0.0.0` is nonsensical and breaks subscribers.

F [... cut]
closed 2022-10-13 by foxymiles