foxglove/studio #4601: the self-host web app crashed after i drag my extension panel to web App while i using dockerfile

hql420621, 2022-10-09:
**Description**

- Version: latest
- OS: linux
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): 
- ROS distro (if applicable):

**Steps To Reproduce**

step 1 ,run a docker image by using `docker run --rm -p "8080:8080" ghcr.io/foxglove/studio:latest`

step 2. open chrome then access the server 

[image]

step3  drag my custom extension into web app,then it crashed!

**Expected Behavior**
the web app can install custom extension correctly ,

**Actual Behavior**
[image]

is there another way to install my custom extension while i use self-host docker excepting drag
can i specify the location where i install my custom extension ,does its stored in the cache,if so ,it will missing after i clear browser cache?

foxymiles, 2022-10-10:
So far I am not able to reproduce this. Does your extension work if you try to install it on https://studio.foxglove.dev?

hql420621, 2022-10-11:
> So far I am not able to reproduce this. Does your extension work if you try to install it on https://studio.foxglove.dev?

yes,it works if i install it on https://studio.foglove.dev? but not if i use docker in self-host way, and is there another way to install extension package into web app except drag the extension?

foxymiles, 2022-10-11:
Could you please try this again in the docker hosted version but this time have the chrome developer tools open and report any error you see there when the installation fails?

So far I am not able to reproduce this locally.

hql420621, 2022-10-12:
this is what developer tools report as below
[image]

LJL0511, 2022-10-12:
I've had this problem. Studio must be deployed behind an https domain.

defunctzombie, 2022-10-13:
> way to install extension package into web app except drag the extension?

The other way we support installing extensions is via our automatic extension provisioning when you are signed into your Foxglove account. This requires using the hosted version of Studio or the Foxglove packaged Studio app.

For open-source builds the only supported workflow for installing extensions is drag/drop.

I am able to reproduce this failure if I run the docker image and open the page using my <ip>:8080. If instead I open `localhost:8080` I am able to drag the extension file without issue. This leads me to think this is something specific to APIs we use to manage extensions that require running in a secure context (https). This works on localhost because browsers often have relaxed the restrictions for localhost to facilitate dev work.

defunctzombie, 2022-10-13:
The crash occurs when this line of code executes: https://github.com/foxglove/studio/blob/main/packages/studio-base/src/components/DocumentDropListener.tsx#L53

commit 23bbfb09 (2022-10-13): Avoid calling `getAsFileSystemHandle` outside of a secure context

The `getAsFileSystemHandle` api is only available in a secure context. Calling it outside of a secure context crashes the tab and is un-catchable. This change adds a check to verify there is a secure context before trying to use the api.

Fixes: #4601
PR #4630 (2022-10-13): Avoid calling `getAsFileSystemHandle` outside of a secure context

**User-Facing Changes**
Dragging a file into studio does not crash the browser when viewing Studio from an unsecure context.

**Description**
The `getAsFileSystemHandle` api is only available in a secure context. Calling it outside of a secure context crashes the tab and is un-catchable. This change adds a check to verify there is a secure context before trying to use the api.

Fixes: #4601
commit e799cea7 (2022-10-13): Avoid calling `getAsFileSystemHandle` outside of a secure context (#4630)

The `getAsFileSystemHandle` api is only available in a secure context.
Calling it outside of a secure context crashes the tab and is
un-catchable. This change adds a check to verify there is a secure
context before trying to use the api.

Fixes: #4601
closed 2022-10-13 by defunctzombie