foxglove/studio #4603: Plot panel reorder topics by dragging

Combinacijus, 2022-10-09:
I would like to reorder topics in Plot panel but can't find how
[image]

jtbandes, 2022-10-12:
This is a good request. I would add that we are more likely to implement it in the sidebar settings (https://github.com/foxglove/studio/pull/4321), since we are exploring whether to replace the editable legend with this sidebar. cc @2metres @foxymiles

foxymiles, 2022-10-12:
Yeah I agree if we're going to do something like this it would make sense as part of the migration to the sidebar. Then other panels could benefit from the work.

jtbandes, 2022-10-12:
We've also already implemented this kind of thing in the sidebar in the Indicator panel (although doesn't support drag-and-drop). It would be nice if this were an inherent feature provided by the sidebar settings API in a reusable way:

[image]