foxglove/studio #4615: Style uplift on Studio Readme

2metres, 2022-10-11:
Feedback from @amacneil:
> Really nice readme hero image, badges and links. we could do something similar for mcap + studio repos: https://github.com/prisma/prisma#readme

[image]

2metres, 2022-10-11:
Might be worth reconsidering the content of the Readme at this momemnt as well

amacneil, 2022-10-12:
Don't overthink it, just a nice header image, centered title, and links is a great improvement.

ocin-rye, 2022-10-13:
Uploading header image here to reference GitHub image URL in the readme file.

[image]

closed 2022-10-24 by ocin-rye