foxglove/studio #4618: 3D Panel can cause page freeze

KisonHe, 2022-10-12:
**Description**
When I connect foxglove studio to my robot the page instantly freeze. So I record a bag file and it still doesn't work. However my rviz is working normally (https://github.com/KisonHe/bug-report-bag-files Got a screenshot)

So I made a ROS bag file and used https://studio.foxglove.dev/ to open it, still it freezes. I am not good at js stuff but I think the 3D panel should not cause browser freeze in any case. I tried opening on many devices including iPad/Chrome, windows/edge,chrome they all freeze. The bag file is here https://github.com/KisonHe/bug-report-bag-files 

After some try i think it might be caused by the tf topic, not sure though

- Version:
- OS: Many OS tried. Manjaro linux, ubuntu ,windows, iPadOS 's browers. All freeze
- Data source bag file at https://github.com/KisonHe/bug-report-bag-files 
- ROS distro (if applicable): melodic

**Steps To Reproduce**

Download the bag file from https://github.com/KisonHe/bug-report-bag-files and open it with foxglove studio.(With a 3d panel)

**Expected Behavior**
It should not freeze

**Actual Behavior**
It freeze

Thx, have a good day :)

defunctzombie, 2022-10-12:
Thank you for the bug report and bag file to reproduce. I can confirm I am able to reproduce with the latest version of studio web. We'll take a look and get this fixed.

snosenzo, 2022-10-12:
Hey so, It looks like the tfMessages in your bag  (see image). 
[image]
they aren’t formatted in the bag as we expect them to be.

the rotation and translation are on the baseTf message instead of being under tfMessage.transforms[0].transform.translation/rotation as our types specify:
```

export type Transform = {
  translation: Vector3;
  rotation: Quaternion;
};

export type TransformStamped = {
  header: Header;
  child_frame_id: string;
  transform: Transform;
};

export type TFMessage = { transforms: TransformStamped[] };

```
and what we have is congruent with https://docs.ros.org/en/noetic/api/tf2_msgs/html/msg/TFMessage.html
So I'm going to work on fixing the crash so that the panel doesn't freeze, but in the meantime if you change your bag to follow this format it should work. 
Also can we ask where this tfMessage comes from? if there's enough demand we could support it, we just haven't seen this before

snosenzo, 2022-10-12:
So it turns out from further investigation that the root cause of the freeze is actually a cyclical transform tree created from the tfMessages. It looks like there's a cycle between the `base_link`, `odom` and `map` coordinate frames in the bag file.

KisonHe, 2022-10-13:
> So it turns out from further investigation that the root cause of the freeze is actually a cyclical transform tree created from the tfMessages. It looks like there's a cycle between the `base_link`, `odom` and `map` coordinate frames in the bag file.

Thanks for your investigation :) Will check about tfMessages and try with fixed ones to see if it still freezes :)

commit 16bd1dae (2022-10-13): 3D Panel can cause page freeze
Fixes #4618
PR #4621 (2022-10-13): 3D Panel will error on cyclical transforms within the TransformTree
**User-Facing Changes**
 - 3D panel will throw an error when a file attempts to add a cyclical transform to the tree. The error tells them what parent/child frame pair first caused the cycle.
 - user must fix file in order to load transforms

**Description**
 - cycles in the transform tree will throw an error instead of freezing the chrome tab.
[image]

Fixes #4618
commit a1a199da (2022-10-13): 3D Panel will error on cyclical transforms within the TransformTree (#4621)

**User-Facing Changes**
- 3D panel will throw an error when a file attempts to add a cyclical
transform to the tree. The error tells them what parent/child frame pair
first caused the cycle.
 - user must fix file in order to load transforms

**Description**
- cycles in the transform tree will throw an error instead of freezing
the chrome tab.
<img width="959" alt="image"
src="https://user-images.githubusercontent.com/10187776/195429903-49b7b92f-5777-47c9-94cd-d8bc2690e1ef.png">

Fixes #4618
commit 92f886a5 (2022-10-13): 3D Panel will error on cyclical transforms within the TransformTree (#4621)

**User-Facing Changes**
- 3D panel will throw an error when a file attempts to add a cyclical
transform to the tree. The error tells them what parent/child frame pair
first caused the cycle.
 - user must fix file in order to load transforms

**Description**
- cycles in the transform tree will throw an error instead of freezing
the chrome tab.
<img width="959" alt="image"
src="https://user-images.githubusercontent.com/10187776/195429903-49b7b92f-5777-47c9-94cd-d8bc2690e1ef.png">

Fixes #4618
closed 2022-10-13 by snosenzo