foxglove/studio #4624: Rendering `foxglove.Grid` is slower than `nav_msgs/OccupancyGrid`

amacneil, 2022-10-12:
**Description**
Anecdotal evidence is that `foxglove.Grid` is slower to render than ROS `nav_msgs/OccupancyGrid`, at least when using protobuf encoding.

- Do we have a way to benchmark this?

- How does ROS 1 `foxglove.Grid` performance compare to `nav_msgs/OccupancyGrid`?

amacneil, 2022-10-12:
cc @james-rms @jtbandes @defunctzombie

jtbandes, 2022-10-12:
To summarize the performance issues and planned work:

- Decoding of Protobuf data is inherently slower than ROS 1 data, because the protobuf [wire format](https://developers.google.com/protocol-buffers/docs/encoding) uses a tag-(length-)value encoding for all field values. We could explore changing the sample data to use flatbuffers instead of protobuf to reduce this extra overhead, but that would be blocked by https://github.com/foxglove/schemas/issues/52. This is mitigated by #4538, as long as we don't end up reverting it.
  - It's also _possible_ that decoding performance is limited by some implementation details of [protobuf.js](https://github.com/protobufjs/protobuf.js). We could also embark on a detailed performance analysis of protobuf.js, and possibly end up making changes to it or making our own protobuf decoder. However, I don't think that would be a good use of our time right now.
  
- I removed the re-publishing of the map every 2 sec in https://github.com/foxglove/nuscenes2mcap/pull/22, because the lookback now makes it behave correctly when seeking.
- OccupancyGrids use `int8[]` data with a [special value→rgba mapping logic](https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ThreeDeeRender/renderables/OccupancyGrids.ts#L293-L344). We can't currently replicate the exact same rendering with a simple `int8` or `uint8` field in a foxglove.Grid — we can get close with a custom gradient, but the OccupancyGrid renderer turns a value of **exactly 100** into a fully transparent cell (while all other values are fully opaque), which isn't something you can do in our custom gradient settings. That's why I opted for using a `uint32` value to encode complete rgba values in the grid in https://github.com/foxglove/nuscenes2mcap/pull/21. This increases the cost of ingesting the grid topic in the sample dataset, because we are processing 4x as many bytes. However, once the color values are uploaded to the GPU they are rendered directly with no perf cost.
  - We could add an "OccupancyGrid" mode to the Grid color settings, which would use the same color mapping function, but I would find that a bit strange and arbitrary for our non-ROS users to see in the Grid settings, and it's not strictly required because there is already have an RGBA color option for ultimate flexibility.
- What @snosenzo is working on is moving the PackedElementField deserialization and the value→rgba conversion off the CPU and into the grid/pointcloud shaders. This will massively improve the ingestion time at the cost of a bit more per-frame GPU work (highly parallel though). (The dream is that we could copy the raw buffers directly to the GPU without doing any extra CPU work, although there are tradeoffs where if grids/pointclouds have a huge number of fields we would be wasting lots of GPU memory by doing this.)

jtbandes, 2022-10-12:
Filed 2 new tickets to track planned work:

- https://github.com/foxglove/studio/issues/4626
- https://github.com/foxglove/studio/issues/4627