foxglove/studio #4625: Missing sample image for "Parameters" panel in new layout view

banisadr, 2022-10-12:
**Description**
Very minor, as described in title - just need to create art & add

**Steps To Reproduce**
Go to studio.foxglove.dev, create new layout

**Expected Behavior**
All panel preview tiles in the new layout landing page should have sample images.

**Actual Behavior**
[image]

defunctzombie, 2022-10-12:
What type of image would speak "parameters" to you?

banisadr, 2022-10-15:
Something marginally less bland than a grey void. My is something similar to the top half of the Data Source Info thumbnail image crossed with this random list of params I just pulled from the ROS Docs?

[image]

defunctzombie, 2022-10-15:
@2metres sending your way to scope and resolve. I agree that every built-in panel should have a thumbnail.

jtbandes, 2022-10-15:
While we’re at it maybe we can get light mode versions of the thumbnails 😀

defunctzombie, 2022-10-15:
> While we’re at it maybe we can get light mode versions of the thumbnails

Scope creep. This is how it starts.

PR #4709 (2022-10-25): Add thumbnail to parameters panel
**User-Facing Changes**
[image]

Closes #4625
commit 3089eada (2022-10-25): Add thumbnail to parameters panel (#4709)

**User-Facing Changes**
<img width="140" alt="thumbnail"
src="https://user-images.githubusercontent.com/924528/197899709-89645b68-1d3c-4ff4-b07a-c0610cd396a2.png">

Closes #4625
closed 2022-10-25 by 2metres