foxglove/studio #4641: Grid z-fighting is annoying to work around

james-rms, 2022-10-14:
**Description**
Currently when generating demo data from NuScenes datasets, we place the driveable area grid 1cm above the map grid. Sometimes, z-fighting still occurs between these two planes:
https://user-images.githubusercontent.com/18162835/195738181-64b06ce8-e1d8-45f1-92c2-704c81d00eed.mov

Whether you see z-fighting depends on:
1. the distance between planes
2. the Z-buffer bit depth
3. the distance of the camera from the planes

Indeed when you zoom in close to the car, the flickering goes away.

I believe this represents a usability issue for users trying to make their visualisation messages look right. To compensate for this now, they have to modify their source data - this is painful if you have a long-running pipeline producing the source data.

We might be able to solve this by documenting a known "safe distance" between planes to prevent z-fighting, based on whatever maximum zoom level we expect to place the camera at. I don't think we have a defined max zoom level though. Also, if this safe distance is too far (say 1m) it prevents users from actually representing their environment accurately in visualization.

One proposed solution is to add a "polygon offset" topic setting to studio, so that users can tweak this offset manually until z-fighting goes away. I think this might be a nice feature, but there would still be a usability problem, because they have no way to fix the experience for all users at once, or for all source data at once.

- Version: studio.foxglove.dev 
- OS: 
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Check the demo data here https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_SUQ41s35DqPfGAoo&ds.end=2018-11-21T11%3A47%3A47.447Z&ds.importId=GtECnrokPxX8s4XM&ds.start=2018-11-21T11%3A47%3A27.948Z&layoutId=76bde952-b9e4-4eaf-be25-6c901e7fc099
- ROS distro (if applicable):

jtbandes, 2022-10-14:
Reading material about polygon offset:
- https://sites.google.com/site/threejstuts/home/polygon_offset
- https://www.opengl.org/archives/resources/faq/technical/polygonoffset.htm

jhurliman, 2022-10-25:
Fixed by https://github.com/foxglove/studio/pull/4664

closed 2022-10-25 by jhurliman

jtbandes, 2022-10-25:
All the issues https://github.com/foxglove/studio/pull/4662 is related to are now closed. Should we abandon that feature then?

jhurliman, 2022-10-25:
I propose significantly deprioritizing it. Logarithmic depth buffers would be useful for rendering objects with different magnitudes of position/scale in the same scene, but this doesn't typically describe robotics visualization and we haven't received a user request for it or bug report that would be addressed by it.