foxglove/studio #4646: Color Point Cloud with 3 uint8s

banisadr, 2022-10-14:
Assigning per-point color in a point cloud is only supported via a single `rgb` or `rgba` unint32. We should also add support for separate `r` `g` `b` channels as 3 unint8s since this is how some customers describe color. Currently those user must perform  an extra step of packing their individual channels into a single unint32.

Desired PackedElementField for a custom colored point cloud with individual color channels:
```
  const std::vector<std::pair<std::string, foxglove::PackedElementField::NumericType>> fields = {
      {"x", fpe::FLOAT32},
      {"y", fpe::FLOAT32},
      {"z", fpe::FLOAT32},
      {"r", fpe::UINT8},
      {"g", fpe::UINT8},
      {"b", fpe::UINT8},
      };
```

danielgordon10, 2022-11-08:
I think https://github.com/foxglove/studio/pull/4762 closes this bug

jtbandes, 2022-11-08:
Yep, it does, thanks!

Technically not as written because we only support 4 uint8s (including alpha) but it satisfies the spirit of the issue. If supporting 3 as well as 4 channels is important to people we can open a new issue.

closed 2022-11-08 by jtbandes

danielgordon10, 2022-11-09:
It would be nice to not have to send a dummy byte and it would make mcaps 6% smaller. As I understand you can already set  opacity for the whole layer in the console viewer which is all I would really want.