foxglove/studio #4647: diagonal and blocky artifacts with overlapping foxglove.Grid instances

james-rms, 2022-10-17:
**Description**
A continuation of #4641 - I tried shortening the far clipping plane of the camera by a factor of 50 (which should increase the bit depth of the Z-buffer by the same factor) but this made no improvement on the number of block artifacts that I previously attributed to z-fighting. There are also some diagonal lines appearing in the artifacts which shouldn't happen as a result of Z-fighting. Some screenshots:

[image]
[image]

- Version: studio.foxglove.dev
- OS: Chrome Version 106.0.5249.103 (Official Build) (arm64)
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): check the NuScenes sample data for a good example
- ROS distro (if applicable):

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

snosenzo, 2022-10-17:
Would a possible solution to this to be to change the example file to put the driveable_area on a coordinate frame as a child of `map` that has a translation like .05 above the map? (that way it can also be adjustable to users)

jtbandes, 2022-10-18:
I played with this a bit more and I saw that adjusting the near plane seemed to have the desired effect of removing the artifacts. So I do believe that we are seeing "regular" z-fighting here. I'm making a PR to allow enabling logarithmic depth: #4662

https://user-images.githubusercontent.com/14237/196517231-d46ffcbf-cf64-491b-9389-ccad8302b5bd.mov

commit 48df4456 (2022-10-18): Tweak NuScenes demo layout near plane (#4664)

This pushes the near plane out to 1 meter, resulting in significantly
better Z-fighting behavior for close planes.

Having a near plane at 1m means one only sees clipping at pretty tight
distances, as shown here:
<img width="1006" alt="Screen Shot 2022-10-19 at 8 05 27 am"
src="https://user-images.githubusercontent.com/18162835/196544799-5565d090-e166-452f-aee2-72b8897d3c88.png">

This may be inappropriate for applications like robot manipulation, but
the setting is still configurable for those situations.

Fixes #4647
closed 2022-10-18 by james-rms