foxglove/studio #4652: Empty "Selected object" inspector popup is unnecessary

amacneil, 2022-10-17:
**Description**
User feedback: when clicking the cursor button to enter object inspect mode (in the 3D panel and Image panel) results in this popup on your screen, even when no object is selected.

[image]

These instructions and popup aren't really necessary: the cursor button turning blue would be sufficient to indicate you are in Inspect mode.

amacneil, 2022-10-17:
cc @2metres @banisadr for feedback

banisadr, 2022-10-17:
I disagree that the pop-up and instructions aren't necessary. For a new user, it is not at all obvious what the arrow does, what object inspector is, or how to use them - the pop-up and instructions solve that.

That said, yes, the empty pop-up is annoying, particularly if using a small panel:
[image]

With a larger panel, however, It's not really in the way and is a good reminder that you're in inspector mode, even for an experienced user:
[image]

I would suggest we keep the default behavior but allow the users to collapse or dismiss the empty pop-up.

defunctzombie, 2022-10-17:
Proposing a middle ground between being able to keep the selection mode on and the dialog hidden when nothing is selected:

- When a user toggles the pointer we open the popup to display the "Click an object" help text. We can make the help text go away after a few moments.
- When a user de-selects an object we hide the popup but keep the selection mode enabled.
- When a user clicks the "collapse" icon and an object is selected, we unselect the object and hide the dialog (but keep the selection mode on).
- When a user clicks another object we re-show the popup.
- When a user clicks the pointer when selection mode is enabled we turn off selection mode.

amacneil, 2022-10-23:
@defunctzombie I think it is potentially confusing if clicking the collapse icon keeps it in selection mode.

Maybe we just show the popup when you first enter selection mode, and dismiss it as soon as you click on empty space? (clicking on "something" will cause the popup to populate with data, clicking on empty space will hide it - at least in 3d panel).

defunctzombie, 2022-10-23:
Proposed operation mode per @amacneil feedback:

- When a user toggles the pointer from _off_ to _on_ we open the popup to display the "Click an object" help text. We can make the help text go away after a few moments.
- When a user de-selects an object we hide the popup but keep the selection mode enabled.
- When a user clicks the "collapse" icon, we unselect the object (if any is selected), hide the dialog, and disable selection mode.
- When a user clicks the pointer when selection mode is enabled we turn off selection mode.

amacneil, 2022-10-26:
Feedback from product review: We should huddle about our plan for a global object inspector, rather than making further changes to this feature that we are planning to change anyway.

defunctzombie, 2022-10-27:
> We should huddle about our plan for a global object inspector, rather than making further changes to this feature that we are planning to change anyway.

My take is that this delays a "good enough" improvement today for a "perfect" change in some future. There's a basic improvement we can make with less overhaul that hides the popup when you de-select an object but still keeps the help text the first time you enter the selection mode.

amacneil, 2022-10-27:
That is why "we should huddle about it". The proposal here adds complexity, adds further deviation between 3d panel and image panel inspector, and the current inspector covers up only a very small portion of the 3d panel. We should discuss it live though, I think there is more nuance than we can discuss here.

amacneil, 2022-10-27:
If we just want to "hide the popup text after you deselect" and that is easy to implement, I'm not against doing it (we should do in image panel too). Once we start getting into "make the help text go away after a few moments" it feels like we are adding complexity. But either way we should discuss what our longer term plan is for this feature.