foxglove/studio #4657: Re-enable transform preloading in 3D panel

jhurliman, 2022-10-17:
Context: https://github.com/foxglove/studio/pull/4656

defunctzombie, 2022-10-18:
Escalating to a bug so we can discuss. This is a feature we've shipped and also started to inform customers about. We need to think through how to bring this back ASAP. I don't think we've updated documentation for transform preloading details yet so we are ok on that front but generally a revert of a supported workflow needs escalated follow-up so we make sure we cleanup docs or other references.

defunctzombie, 2022-10-19:
When we re-enable using preloaded transforms we need to ensure we avoid this bug: https://github.com/foxglove/studio/issues/4605

defunctzombie, 2022-12-12:
Closing this out. We've created a linear project tracking adding this back in and have prioritized it. I consider us tackling this high priority and we've noted that in the planning of our projects.

closed 2022-12-12 by defunctzombie