foxglove/studio #4667: Text Input on Settings Gradient Picker doesn't submit on blur or return key

2metres, 2022-10-19:
**Description**
Text Input on Settings Gradient Picker doesn't submit on blur or return key, Observed in the Indicator panel

- Version:
- OS: Web, MacOS, Chrome 106
- Data source: Demo Bag

**Steps To Reproduce**
[
](https://user-images.githubusercontent.com/924528/196569543-7d63e196-780d-4a90-9266-c056d2fcf952.mov)

**Expected Behavior**
Blur or enter should set value

osaro-imohe, 2022-10-20:
I'll get this

jtbandes, 2022-11-08:
@osaro-imohe Since there's been no update on this issue I'm going to remove your assignment for now

snosenzo, 2022-11-14:
Able to only reproduce partially. Maybe this behavior has changed in the past month?
This is the experience that I am seeing:
 - On changing the value in the gradient picker, the value in the settings is immediately reflective of the change. The gradient picker remains open while the value in the settings is changing.
 - Closing of the gradient picker onBlur is working, but on enter it is not closing (can address in PR)
 - changing the value via the textbox in the gradient picker only updates the settings, but does not update the panel (will address as part of PR). Clicking and dragging over the colorspace to choose a color fixes this.

Let me know if this aligns with your current experience

PR #4810 (2022-11-14): Fix color picker text input not updating values
**User-Facing Changes**
 - Fix color picker text input not updating panels

**Description**
- close color picker on enter key text input
- fix text input values being invalid for consumption by panels by adding `#` to beginning of value string for consumption elsewhere in UI

Fixes: #4667
commit 239738c9 (2022-11-14): Fix color picker text input not updating values (#4810)

**User-Facing Changes**
 - Fix color picker text input not updating panels

**Description**
- close color picker on enter key text input
- fix text input values being invalid for consumption by panels by
adding `#` to beginning of value string for consumption elsewhere in UI

Fixes: #4667
closed 2022-11-14 by jtbandes