foxglove/studio #4670: Map panel should use 👁️ visibility toggles rather than On/Off switch

jtbandes, 2022-10-19:
Map panel:

[image]

3D panel:
[image]

Plot panel:

[image]

defunctzombie, 2022-10-19:
@foxymiles you did the initial settings port: https://github.com/foxglove/studio/commit/4e156abb8fb84076d1ccacefb2b45731c730414b

Do you remember why the toggle was picked vs the visibility eye?

foxymiles, 2022-10-19:
I think it was just because this was one of the very first panels we ported and we hadn't really settled on conventions like this yet.

osaro-imohe, 2022-10-20:
I'll get this