foxglove/studio #4672: Data source info sidebar should show start/end time in RFC 3339 format

defunctzombie, 2022-10-19:
Our APIs and CLI tools accept times in RFC 3339 format. To make it easier to copy/paste timestamps into the tools we should display these times in the same format.

[image]

jtbandes, 2022-10-19:
I would argue that the RFC 3339 format is less “human readable”. Maybe we need a user setting so we can keep the current behavior around as an option, or we need a special copy button/dropdown that lets you choose a specific format when copying?

defunctzombie, 2022-10-20:
> I would argue that the RFC 3339 format is less “human readable”. Maybe we need a user setting so we can keep the current behavior around as an option, or we need a special copy button/dropdown that lets you choose a specific format when copying?

I'd say its a thin argument. I guess the `T` is a bit strange and doesn't help break up the date and time visually.

I could add another time format option to settings but generally I am leaning towards it being better to have time values in our apps you can copy/paste to other tools. Right now it is an immense struggle to go from having some times in Studio to using those in our CLI.

jtbandes, 2022-10-20:
Yep, I am definitely +1 on making it easier to copy/paste to other tools. But I do think `2022-10-20T20:48:41.069Z` is harder to visually parse than `2022-10-20 8:48:41.069 PM UTC`.

defunctzombie, 2022-10-21:
Other output from some tools we use:

```
(ros_env) ➜  Downloads mcap info output.mcap
library: libmcap 0.1.1
profile: protobuf
messages: 1
duration: 0s
start: 2022-06-10T07:15:45.430695138+09:00 (1654812945.430695138)
end: 2022-06-10T07:15:45.430695138+09:00 (1654812945.430695138)
```

```
version:      2.0
duration:     25:11s (1511s)
start:        Jun 25 2022 12:02:23.85 (1656126143.85)
end:          Jun 25 2022 12:27:35.46 (1656127655.46)
size:         504.0 MB
```

Not much consistency today ♾️ . Studio !== mcap !== rosbag :(

defunctzombie, 2022-11-15:
Closing this as not planned for now. I think what we have is poor usability but don't have the time to tackle this. I won't forget about it and might come up with newer ways to resolve this.

closed 2022-11-15 by defunctzombie