foxglove/studio #4680: RGB Point cloud color is wrong

defunctzombie, 2022-10-20:
**Description**

- Open [this bag file][1] in Studio (web or desktop).
- Open the 3d panel and enable the only point cloud topic
- Notice the point cloud is red (sometimes).
- If the cloud is not red, change the color to flag then back to RGB.

Here is a screenshot showing the same point cloud in Studio and RViz.

[image]

---
- Version: 1.29.0
- OS: Mac / Web / Chrome
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag
- ROS distro (if applicable): Noetic

[1]: https://drive.google.com/file/d/1RAxFlp4IEncfvcA1wdyNTOQYgUE7XJYH/view?usp=sharing

amacneil, 2022-10-20:
Shouldn't we use the same color byte order as rviz by default, when rendering `PointCloud2`?

defunctzombie, 2022-10-21:
Re-opened this bug since this did work in legacy 3d. Tho it might be considered a duplicate of https://github.com/foxglove/studio/issues/4483?

jtbandes, 2022-10-21:
Duplicate of #4483

PR #4685 (2022-10-20): [3D] fix auto-selection of rgbByteOrder
**User-Facing Changes**
[3D] Fixed a bug with RGB point clouds where the automatically-selected settings would sometimes change after the point cloud was initially displayed.

**Description**
Fixes #4680

We were using the `autoSelectColorField` function to choose the colorMode and rgbByteOrder, but the rgbByteOrder wasn't actually being saved in the panel config. So when the renderable got updated, it would re-read the saved config value, which was undefined, and fall back to a different value.
commit a2f9fbd9 (2022-10-20): [3D] fix auto-selection of rgbByteOrder (#4685)

**User-Facing Changes**
[3D] Fixed a bug with RGB point clouds where the automatically-selected
settings would sometimes change after the point cloud was initially
displayed.

**Description**
Fixes #4680

We were using the `autoSelectColorField` function to choose the
colorMode and rgbByteOrder, but the rgbByteOrder wasn't actually being
saved in the panel config. So when the renderable got updated, it would
re-read the saved config value, which was undefined, and fall back to a
different value.
closed 2022-10-21 by jtbandes